ViperRNMC/marstek_venus_modbus #25: Modbus Gateway

mathias-goebel, 2025-08-17:
I can confirm the Integration is working with a different Gateway, the PUSR DR134 providing an Ethernet port.

Maybe further testing is needed, but I do not see any reason the integration should not work with any ModbusTCP gateway. Would you mind adding a list of supported or tested Gateways to operate with?

Waldo56, 2025-08-30:
Works on the Waveshare RS485 to RJ45 Ethernet Converter also.

Perfect work and many thanks.

Stiopa21, 2025-11-28:
Bonjour et merci pour votre réponse, Voici mon problème Je n'arrive pas à avoir des remontée d'information. Je pense que mon elfin est bon. 

[image]

ViperRNMC, 2025-11-28:
Do you install the ew11 correct? See details here https://github.com/ViperRNMC/marstek_venus_modbus/blob/main/device/elfin_ew11a.md

Stiopa21, 2025-11-28:
Oui, j'avais vue votre paramétrage du elfin, je vais essayer d'inverser les fils data pour voir,

Stiopa21, 2025-11-29:
est ce que dans le LAN de Elfin je met l'adresse de Home Assistant ?

Stiopa21, 2025-11-29:
quand j'active rs485 le relais de la batterie ne claque pas

Stiopa21, 2025-11-29:
👏🏻👍🏻🎉🎁🎈🎉🎊💪🏻 merci, et bravo pour votre travail 

[image]

Stiopa21, 2025-11-30:
sa fonctionne vraiment bien c'est super merci pour votre travail, j'ai une question pour un reglage du niveau de SOC max comment faire svp entre charge SOC ou charging cutoff je n'ai pas bien compris.

ViperRNMC, 2025-11-30:
The cutoff also limits during normal operation, while the other one is only active during manual control.

Stiopa21, 2025-11-30:
Merci, si je comprends bien il vaut mieux que j'utilise charge SOC pour limiter la charge au niveau choisi

Stiopa21, 2025-11-30:
Merci, je vais faire quelques essais, le but étant d'écrire un yamel qui va limiter la recharge nocturne en fonction des prévisions météo du lendemain