ViperRNMC/marstek_venus_modbus #29: Unavailable after HA upgrade to 2025.9

Peekewee, 2025-09-04:
The integration became unavailable after ha upgrade to 2025.9
In evcc the connection is still working, som modbus and elfin are ok

Peekewee, 2025-09-04:
Following error in logs:

This error originated from a custom integration.

Logger: custom_components.marstek_modbus.helpers.modbus_client
Source: custom_components/marstek_modbus/helpers/modbus_client.py:141
integration: Marstek Venus Modbus (documentation)
First occurred: 09:13:34 (14707 occurrences)
Last logged: 10:12:32

Exception during Modbus read at register 32204 (0x7DCC) on attempt 3: ModbusClientMixin.read_holding_registers() got an unexpected keyword argument 'slave'
Exception during Modbus read at register 33000 (0x80E8) on attempt 1: ModbusClientMixin.read_holding_registers() got an unexpected keyword argument 'slave'
Exception during Modbus read at register 33000 (0x80E8) on attempt 2: ModbusClientMixin.read_holding_registers() got an unexpected keyword argument 'slave'
Exception during Modbus read at register 33000 (0x80E8) on attempt 3: ModbusClientMixin.read_holding_registers() got an unexpected keyword argument 'slave'
Exception during Modbus read at register 33002 (0x80EA) on attempt 1: ModbusClientMixin.read_holding_registers() got an unexpected keyword argument 'slave'
Traceback (most recent call last):
  File "/config/custom_components/marstek_modbus/helpers/modbus_client.py", line 141, in async_read_register
    result = await self.client.read_holding_registers(
                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        address=register, count=count, slave=self.unit_id
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
TypeError: ModbusClientMixin.read_holding_registers() got an unexpected keyword argument 'slave'

RetroButton, 2025-09-04:
Same here. Installed the integration today.
Direct ModBus readout using HA works fine.