ViperRNMC/marstek_venus_modbus #46: Add v3 suppert

ViperRNMC, 2025-09-22:
Need some time to build a list for v3, base can be used for both.

https://docs.google.com/spreadsheets/d/e/2PACX-1vSUZ-IOsrAn26ww4_14yPE9ee0fXafxB7RdMK8pC9Mha590iBD7PjeSpv17L4Ji0oTtRB4w6WG77Iwd/pubhtml?gid=0&single=true

Strilled, 2025-10-26:
I was able to read the AC power via Modbus in HomeAssistant with the following block in the configuration.yml:
```
      - name: "Marstek-VenusE-3-1 AC Power"
        unit_of_measurement: W
        slave: 1
        address: 37004
        input_type: holding
        data_type: int16
        scan_interval: 1
        precision: 0
```
Positive values are displayed when the battery is discharging and negative values when the battery is charging.

Maybe this will help you :)

allteeI, 2025-10-28:
I have a v3, too.
This is what I see, if I select v1/v2:

But: I have a problem with my battery at the moment and are already in contact with marstek support. Currently my battery won’t charge nor discharge. Neither via one of the App integrated work modes nor the force mode. But as you can see most of the registers seem to work. Ok some temperatures have wrong scales but the rest seems ok.

[image]

[image]
[image]
[image]
[image]
[image]

ViperRNMC, 2025-10-29:
Max (dis)charge power indeed limit also the manual (dis)charge. about the scale of cell temp, we also see this in v2 with BMS 213 https://github.com/ViperRNMC/marstek_venus_modbus/pull/21. Thanks, I'll update the register v3 with these info.

trinity2000, 2025-10-29:
Thank you, i will post some Screenshots, the following entities appear by default:

[image]

[image]

[image]

I have set the app to a fixed discharge of 300W. I don't find this value ‘yet’....

ViperRNMC, 2025-10-29:
See TS.

> Please include at least a 2-minute debug log in the post, so that I can be sure all sensors have been requested by the Modbus client. This way, I can check whether the register can actually be requested and what the raw result looks like.
> 
> Also, please indicate which EMS/BMS version you have, as I’ve learned that older versions don’t yet include all Modbus registers.
> 
> Some info found https://docs.google.com/spreadsheets/d/e/2PACX-1vSUZ-IOsrAn26ww4_14yPE9ee0fXafxB7RdMK8pC9Mha590iBD7PjeSpv17L4Ji0oTtRB4w6WG77Iwd/pubhtml?gid=0&single=true

ViperRNMC, 2025-10-29:
Thanks. A few things I notice right away: grid_standard is not correct — I will temporarily disable it. The battery power, current, and AC current values are incorrect; this needs further investigation. Only BMS is in the app.

trinity2000, 2025-10-29:
The registers for AC power were correct in the V1/V2 version. You could transfer them to the V3.

ViperRNMC, 2025-10-29:
I'll change it in the next beta. Can you make a 5min log with only ac/battery/offgrid power/current/voltage enabled en change is to 75watt discharge en like 50 charge in manual?

trinity2000, 2025-10-29:
300 W Charging in man. Charge Mode:

[image]

2025-10-29 17:21:56.745 WARNING (MainThread) [custom_components.marstek_modbus.coordinator] Invalid value for sensor 'ac_offgrid_power': None (type NoneType)
2025-10-29 17:22:06.346 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Coordinator poll tick at 2025-10-29T16:22:06.346609+00:00
2025-10-29 17:22:06.347 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Dependency key 'total_charging_energy'
2025-10-29 17:22:06.347 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Dependency key 'ac_power'
2025-10-29 17:22:06.347 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Dependency key 'battery_power'
2025-10-29 17:22:06.347 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Dependency key 'total_discharging_energy'
2025-10-29 17:22:06.347 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Dependency key 'battery_total_energy'
2025-10-29 17:22:06.348 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Dependency key 'battery_soc'
2025-10-29 17:22:06.348 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Dependency key 'total_monthly_charging_energy'
2025-10-29 17:22:06.348 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Dependency key 'total_monthly_discharging_energy'
2025-10-29 17:22:06.348 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Skipping disabled entity 'Device Name'
2025-10-29 17:22:06.348 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Skipping disabled entity 'Communication Module Firmware'
2025-10-29 17:22:06.348 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Skipping disabled entity 'MAC Address'
2025-10-29 17:22:06.349 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Fetching disabled dependency key 'battery_soc'
2025-10-29 17:22:06.408 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Requesting register 37005 (0x908D) for sensor 'battery_soc' (type: uint16, count: 1)
2025-10-29 17:22:06.409 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Received data from register 37005 (0x908D): [77]
2025-10-29 17:22:06.409 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Updated sensor 'battery_soc': register=37005, value=77, scale=1, unit=%
2025-10-29 17:22:06.409 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Fetching disabled dependency key 'battery_total_energy'
2025-10-29 17:22:06.468 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Requesting register 32105 (0x7D69) for sensor 'battery_total_energy' (type: uint16, count: 1)
2025-10-29 17:22:06.468 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Received data from register 32105 (0x7D69): [5120]
2025-10-29 17:22:06.469 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Updated sensor 'battery_total_energy': register=32105, value=5120, scale=0.001, unit=kWh
2025-10-29 17:22:06.528 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Requesting register 30100 (0x7594) for sensor 'battery_voltage' (type: uint16, count: 1)
2025-10-29 17:22:06.528 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Received data from register 30100 (0x7594): [5320]
2025-10-29 17:22:06.529 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Updated sensor 'battery_voltage': register=30100, value=5320, scale=0.01, unit=V
2025-10-29 17:22:06.589 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Requesting register 32101 (0x7D65) for sensor 'battery_current' (type: int16, count: 1)
2025-10-29 17:22:06.589 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Received data from register 32101 (0x7D65): [4]
2025-10-29 17:22:06.589 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Updated sensor 'battery_current': register=32101, value=4, scale=0.01, unit=A
2025-10-29 17:22:06.684 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Requesting register 30001 (0x7531) for sensor 'battery_power' (type: int32, count: 2)
2025-10-29 17:22:06.684 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Received data from register 30001 (0x7531): [267, 327]
2025-10-29 17:22:06.685 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Updated sensor 'battery_power': register=30001, value=17498439, scale=1, unit=W
2025-10-29 17:22:06.685 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Skipping disabled entity 'Internal Temperature'
2025-10-29 17:22:06.685 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Skipping disabled entity 'Internal MOS1 Temperature'
2025-10-29 17:22:06.685 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Skipping disabled entity 'Internal MOS2 Temperature'
2025-10-29 17:22:06.748 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Requesting register 32200 (0x7DC8) for sensor 'ac_voltage' (type: uint16, count: 1)
2025-10-29 17:22:06.748 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Received data from register 32200 (0x7DC8): [2262]
[... cut ...]
2025-10-29 17:24:00.741 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Skipping disabled entity 'Charge to SOC'
2025-10-29 17:24:00.741 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Skipping disabled entity 'Backup Function'
2025-10-29 17:24:00.741 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Skipping disabled entity 'RS485 Control Mode'
2025-10-29 17:24:00.741 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Finished fetching MarstekCoordinator data in 0.396 seconds (success: True)

allteeI, 2025-11-01:
[scan2025-11-01_13-29-12.csv](https://github.com/user-attachments/files/23283088/scan2025-11-01_13-29-12.csv)

I did a brute-force scan on 30000-49999 registers, maybe it helps.

ViperRNMC, 2025-11-03:
> [scan2025-11-01_13-29-12.csv](https://github.com/user-attachments/files/23283088/scan2025-11-01_13-29-12.csv)
> 
> I did a brute-force scan on 30000-49999 registers, maybe it helps.

thanks, can you share you bms version?

allteeI, 2025-11-03:
> > [scan2025-11-01_13-29-12.csv](https://github.com/user-attachments/files/23283088/scan2025-11-01_13-29-12.csv)
> > I did a brute-force scan on 30000-49999 registers, maybe it helps.
> 
> thanks, can you share you bms version?

When I made the above attached scan I was on BMS 104.

I've update to 106 during the weekend, if you want me to, I can rescan this evening on BMS 106.

I was also able to confirm the registers holding all the single cell voltages. I don't remember the register addresses right now, but I have it at home (I'm currently at work). --> If you want to implement them as well in the integration?

ViperRNMC, 2025-11-03:
> > > In the attachment you can find 2 log files. Both with the latest bete varsion. One is the V3 configured as a V1/V2. The other one is the V3 configured as a V3. If you need more, just ask.
> > > Ps. Where can we buy you a coffee for your time and effort?
> > > Kind regards, Jonas
> > > [home-assistant_marstek_modbus_2025-10-31T13-38-33.548Z_Beta_ConfiguredAsV2.log](https://github.com/user-attachments/files/23263189/home-assistant_marstek_modbus_2025-10-31T13-38-33.548Z_Beta_ConfiguredAsV2.log) [home-assistant_marstek_modbus_2025-10-31T13-30-13.869Z_Beta_ConfiguredAsV3.log](https://github.com/user-attachments/files/23263188/home-assistant_marstek_modbus_2025-10-31T13-30-13.869Z_Beta_ConfiguredAsV3.log)
> > 
> > 
> > Thanks, version of BMS EMS you have is 213/155, interesting. What I see is (dis)charging_cutoff_capacity is available. Are you sure you have an Marstek Venus E V3?
> > ```
> > 2025-10-31 14:27:50.036 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Requesting register 44000 (0xABE0) for sensor 'charging_cutoff_capacity' (type: uint16, count: 1)
> > 2025-10-31 14:27:50.036 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Received data from register 44000 (0xABE0): [1000]
> > 2025-10-31 14:27:50.037 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Updated number 'charging_cutoff_capacity': register=44000, value=1000, scale=0.1, unit=%
> > 2025-10-31 14:27:50.152 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Requesting register 44001 (0xABE1) for sensor 'discharging_cutoff_capacity' (type: uint16, count: 1)
> > 2025-10-31 14:27:50.152 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Received data from register 44001 (0xABE1): [120]
> > 2025-10-31 14:27:50.152 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Updated number 'discharging_cutoff_capacity': register=44001, value=120, scale=0.1, unit=%  
> > ```
> 
> Hi, Yes, I have both — a V2 (= Marstek 1) running 155/213 and a V3 (Marstek 2) running 139. If you want, I can disable the V2 so you only get logs from a single device to make them easier to read.
> 
> <img alt="Image" width="1661" height="343" src="https://private-user-images.githubusercontent.com/15230807/508954847-3b65f39b-b104-4231-a46a-567e892b4ad1.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjIxODU3NDIsIm5iZiI6MTc2MjE4NTQ0MiwicGF0aCI6Ii8xNTIzMDgwNy81MDg5NTQ4NDctM2I2NWYzOWItYjEwNC00MjMxLWE0NmEtNTY3ZTg5MmI0YWQxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTExMDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUxMTAzVDE1NTcyMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWMxZmFlYmZhMGE5NzFjNzI4Mzk5Njc0Mzg5MTQzNjE5MzA0NTk4NWMyNDYyZWZlZjQzMGI2ZWJhNTYxNTU2ZDUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.cv3TIySetZ_WLqv5OoHKEAEqdDTM-FgU-yqOnqNecl0">

Please disable v2, I can't see difference in the log.

allteeI, 2025-11-04:
Ah ok, I‘ll have a look and we“ll see what I can find out. I thought it might be good to enter timeslots for charge / discharge in manual mode as well before the next scan.

allteeI, 2025-11-04:
this is a manual discharge (set to 432W)
[scan2025-11-04_18-46-31_manual_discharge_423w.csv](https://github.com/user-attachments/files/23344147/scan2025-11-04_18-46-31_manual_discharge_423w.csv)

and a manual charge (set to 123W)
[scan2025-11-04_19-38-57_manual_charge_123w.csv](https://github.com/user-attachments/files/23344150/scan2025-11-04_19-38-57_manual_charge_123w.csv)

cell voltages seems to be correct in new beta. In my HA they all show up as 3V, but when I set accuracy to 0.001 they're correct, I think this is caused by my HA settings and not by the addon (maybe someone else can confirm). Currently I have no time to play around, but I can do these scans (not really time consuming for me, it's a script).

If you want me to scan during specific settings, or specific register ranges just ask.

allteeI, 2025-11-05:
I found following two issues in beta3:

1. as discussed above: cell voltages have wrong precision: showing 3 instead of 3,xxx (single cell voltages 1-16 & min cell voltage & max cell voltage)

2. naming of some entities is inconsistent:
 
I have 3-off Venus 3.0. The first I installed was called "Venus**3**", the second "Venus**2**" and the third I called "Venus**1**"

I expected the naming of the entities in Home Assistant to follow this sheme: <sensor/switch/input>.venus**N**_XXX (with **N** to be the number of my Venus)  
This is right for most of the entities:
examples:
switch.venus**1**_backup_function
sensor.venus**1**_ac_power
...

But: some entities are named different, they seem to follow the sheme: <sensor/switch/input>.marstek_venus_XXX_**N** (with N not being the number of my Venus, here N is an increasing number in the order of my installation)  

now I have the following entities belonging to:

Venus**1** : select.marstek_venus_modbus_force_mode_**3**
Venus**2** : select.marstek_venus_modbus_force_mode_**2**
Venus**3** : select.marstek_venus_modbus_force_mode

I am not sure, if this is caused by previous tries I did without the VIPERRNMC/marstek_venus_modbus Integration. Maybe someone can check if the "Force Mode" entity is named in the same manner as it is in my installation.

If so, there are more entities with the same "wrong" sheme. I can list them, but I think Viper will find this in the code faster than I need to list them.

Grandjude44, 2025-11-10:
Venus E V3 with waveshare rs485 to rj45: 15 entities

mrestorff, 2025-11-18:
I'm also getting some errors on the latest stable release:

```
Logger: custom_components.marstek_modbus.helpers.modbus_client
Quelle: custom_components/marstek_modbus/helpers/modbus_client.py:177
Integration: Marstek Venus Modbus (Dokumentation)
Erstmals aufgetreten: 00:50:42 (8 Vorkommnisse)
Zuletzt protokolliert: 08:53:25

Exception during Modbus read at register 35000 (0x88B8) on attempt 1: Modbus Error: [Input/Output] ERROR: request uses transaction id=2164 but received 0.
Exception during Modbus read at register 30100 (0x7594) on attempt 1: Modbus Error: [Input/Output] ERROR: request uses transaction id=2896 but received 0.
Exception during Modbus read at register 44002 (0xABE2) on attempt 1: Modbus Error: [Input/Output] ERROR: request uses transaction id=5564 but received 0.
Exception during Modbus read at register 35100 (0x891C) on attempt 1: Modbus Error: [Input/Output] ERROR: request uses transaction id=5587 but received 0.
Exception during Modbus read at register 33008 (0x80F0) on attempt 1: Modbus Error: [Input/Output] ERROR: request uses transaction id=7067 but received 0.
Traceback (most recent call last):
  File "/config/custom_components/marstek_modbus/helpers/modbus_client.py", line 177, in async_read_register
    result = await self.client.read_holding_registers(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        address=register, count=count, device_id=self.unit_id
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 184, in execute
     raise ModbusIOException(
         f"ERROR: request uses transaction id={request.transaction_id} but received {response.transaction_id}."
    )
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] ERROR: request uses transaction id=1045 but received 0.
```

as well as

```
Logger: pymodbus.logging
Quelle: runner.py:289
Erstmals aufgetreten: 17. November 2025 um 23:58:22 (28950 Vorkommnisse)
Zuletzt protokolliert: 09:22:00

ERROR: request ask for transaction_id=7240 but got id=7227, Skipping. >>>>> recv: 0x1c 0x3a 0x0 0x0 0x0 0x7 0x1 0x3 0x4 0x0 0x0 0x0 0x0 extra data: >>>>> send: 0x1c 0x3b 0x0 0x0 0x0 0x6 0x1 0x3 0x7d 0x66 0x0 0x2 >>>>> recv: 0x1c 0x3b 0x0 0x0 0x0 0x7 0x1 0x3 0x4 0x0 0x0 0x0 0x0 extra data:
ERROR: request ask for transaction_id=7240 but got id=51, Skipping. >>>>> recv: 0x1c 0x3b 0x0 0x0 0x0 0x7 0x1 0x3 0x4 0x0 0x0 0x0 0x0 extra data: >>>>> recv: 0x0 0x33 0x0 0x0 0x0 0xd 0x1 0x3 0xa 0x9 0x3d 0x9 0x3d 0x0 0x0 0x0 0x0 0x1 0xf4 extra data:
ERROR: request ask for transaction_id=7227 but got id=51, Skipping. >>>>> recv: 0x0 0x33 0x0 0x0 0x0 0xd 0x1 0x3 0xa 0x9 0x3d 0x9 0x3d 0x0 0x0 0x0 0x0 0x1 0xf4 extra data:
ERROR: request ask for transaction_id=7240 but got id=52, Skipping. >>>>> recv: 0x0 0x34 0x0 0x0 0x0 0xd 0x1 0x3 0xa 0x9 0x3e 0x9 0x3e 0x0 0x0 0x0 0x0 0x1 0xf4 extra data:
ERROR: request ask for transaction_id=7227 but got id=52, Skipping. >>>>> recv: 0x0 0x34 0x0 0x0 0x0 0xd 0x1 0x3 0xa 0x9 0x3e 0x9 0x3e 0x0 0x0 0x0 0x0 0x1 0xf4 extra data:
```

This is using the V3 with a PUSR DR134 Modbus RTU to TCP converter.

allteeI, 2025-11-18:
> In the latest loggings from [allteeI](https://github.com/allteeI), 32101 is not current, 32104 is freq. Maybe they are changing the registers? allele can you confirm? and check firmware.

32101 without charging/discharging = 0
32101 while charging @2500W = 46
32101 while discharging @550W = 39312
all values unsigned int
EMS: 139
BMS: 106
VMS: 111

> Other think, can someone with v3 check:
> 
> 44000 : Charging Cutoff Capacity 44001 : Discharging Cutoff Capacity 41010 : Power Restriction Discharge (800/2500 option in app)

44000: illegal data address error (0x02)
44001: illegal data address error (0x02)
41010: illegal data address error (0x02)

is what I get on these registers

mrestorff, 2025-11-18:
Can confirm that cutoff capacity and power restriction registers don't work on V3.

allteeI, 2025-11-23:
as you know I have three Venus E 3.0 and getting more and more trouble with connections. I run a lot in timeouts. my logs have thousands of messages like this

`2025-11-23 10:21:32.449 WARNING (MainThread) [custom_components.marstek_modbus.coordinator] Timeout reading sensor 'inverter_state' at register 35100 - connection may be slow or incorrect`

I am unsure where to look at. My feeling is, that it was better with 2025.10.xx beta versions than with the .11.x but it is only a feeling. But I am even unsure if it is the integration or maybe my settings in the PUSR. I already disabled a lot of the entities, I don't really need, but it did not fix the problem. Polling interval are all set to default values. 

I have a node red logic which controls the charge/discharge. This runs every 15 seconds, which I think should not be a problem, but it seems to me, that it collades even with the regular polling of the entities.

Any ideas where to look at?

Waldo56, 2025-11-23:
Multihost enabled at the RS485 Gateway?

DerAutomatiker, 2025-11-23:
I have something similar but the id's are only off by one not the same as yours.
About 400 occurrences in 24h

Logger: pymodbus.logging
Quelle: runner.py:289
Erstmals aufgetreten: 22. November 2025 um 21:15:40 (417 Vorkommnisse)
Zuletzt protokolliert: 22:53:43

ERROR: request ask for transaction_id=64626 but got id=64625, Skipping. >>>>> send: 0x3b 0x9 0x0 0x0 0x0 0x6 0x1 0x3 0xa4 0x24 0x0 0x1 >>>>> recv: 0x3b 0x9 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x0 extra data: >>>>> send: 0x3b 0xa 0x0 0x0 0x0 0x6 0x1 0x3 0xa4 0x25 0x0 0x1 >>>>> recv: 0x3b 0xa 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x0 extra data: >>>>> send: 0x3b 0xb 0x0 0x0 0x0 0x6 0x1 0x3 0xab 0xe2 0x0 0x1 >>>>> recv: 0x3b 0xb 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x0 extra data: >>>>> send: 0x3b 0xc 0x0 0x0 0x0 0x6 0x1 0x3 0xab 0xe3 0x0 0x1 >>>>> recv: 0x3b 0xc 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x0 extra data: >>>>> send: 0x3b 0xd 0x0 0x0 0x0 0x6 0x1 0x3 0xa4 0x1b 0x0 0x1 >>>>> recv: 0x3b 0xd 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x0 extra data: >>>>> send: 0x3b 0xe 0x0 0x0 0x0 0x6 0x1 0x3 0xa0 0xf0 0x0 0x1 >>>>> recv: 0x3b 0xe 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x1 extra data: >>>>> send: 0x3b 0xf 0x0 0x0 0x0 0x6 0x1 0x3 0xa4 0x10 0x0 0x1 >>>>> recv: 0x3b 0xf 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x55 0xbb extra data: >>>>> send: 0xfc 0x71 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x8 0x0 0x1 >>>>> Repeating.... >>>>> send: 0xfc 0x71 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x8 0x0 0x1 >>>>> recv: 0xfc 0x71 0x0 0x0 0x0 0x5 0x1 0x4 0x2 0x1 0xe9 extra data: >>>>> send: 0xfc 0x72 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x9 0x0 0x1 >>>>> recv: 0xfc 0x71 0x0 0x0 0x0 0x5 0x1 0x4 0x2 0x1 0xe9 0xfc 0x71 0x0 0x0 0x0 0x5 0x1 0x4 0x2 0x1 0xe9 0xfc 0x71 0x0 0x0 0x0 0x5 0x1 0x4 0x2 0x1 0xe9 extra data:
ERROR: request ask for transaction_id=64626 but got id=64625, Skipping.
ERROR: request ask for transaction_id=64908 but got id=64907, Skipping. >>>>> send: 0xfd 0x84 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x12 0x0 0x1 >>>>> recv: 0xfd 0x84 0x0 0x0 0x0 0x5 0x1 0x4 0x2 0x0 0x0 extra data: >>>>> send: 0xfd 0x85 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x77 0x0 0x1 >>>>> recv: 0xfd 0x85 0x0 0x0 0x0 0x5 0x1 0x4 0x2 0x0 0xa extra data: >>>>> send: 0xfd 0x86 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x4 0x0 0x1 >>>>> recv: 0xfd 0x86 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x28 extra data: >>>>> send: 0xfd 0x87 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x5 0x0 0x1 >>>>> recv: 0xfd 0x87 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x1e extra data: >>>>> send: 0xfd 0x88 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x6 0x0 0x1 >>>>> recv: 0xfd 0x88 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0xa extra data: >>>>> send: 0xfd 0x89 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0xc 0x0 0x1 >>>>> recv: 0xfd 0x89 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x3 extra data: >>>>> send: 0xfd 0x8a 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0xd 0x0 0x1 >>>>> recv: 0xfd 0x8a 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x0 extra data: >>>>> send: 0xfd 0x8b 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0xe 0x0 0x1 >>>>> Repeating.... >>>>> send: 0xfd 0x8b 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0xe 0x0 0x1 >>>>> recv: 0xfd 0x8b 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x8 extra data: >>>>> send: 0xfd 0x8c 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0xf 0x0 0x1 >>>>> recv: 0xfd 0x8b 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x8 0xfd 0x8b 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x8 extra data:
ERROR: request ask for transaction_id=64908 but got id=64907, Skipping.
ERROR: request ask for transaction_id=267 but got id=266, Skipping. >>>>> send: 0x3e 0x45 0x0 0x0 0x0 0x6 0x1 0x3 0xa4 0x1a 0x0 0x1 >>>>> recv: 0x3e 0x45 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x0 extra data: >>>>> send: 0x3e 0x46 0x0 0x0 0x0 0x6 0x1 0x3 0xa4 0x24 0x0 0x1 >>>>> recv: 0x3e 0x46 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x0 extra data: >>>>> send: 0x3e 0x47 0x0 0x0 0x0 0x6 0x1 0x3 0xa4 0x25 0x0 0x1 >>>>> recv: 0x3e 0x47 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x0 extra data: >>>>> send: 0x3e 0x48 0x0 0x0 0x0 0x6 0x1 0x3 0xab 0xe2 0x0 0x1 >>>>> recv: 0x3e 0x48 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x0 extra data: >>>>> send: 0x3e 0x49 0x0 0x0 0x0 0x6 0x1 0x3 0xab 0xe3 0x0 0x1 >>>>> recv: 0x3e 0x49 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x0 extra data: >>>>> send: 0x3e 0x4a 0x0 0x0 0x0 0x6 0x1 0x3 0xa4 0x1b 0x0 0x1 >>>>> recv: 0x3e 0x4a 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x0 extra data: >>>>> send: 0x3e 0x4b 0x0 0x0 0x0 0x6 0x1 0x3 0xa0 0xf0 0x0 0x1 >>>>> recv: 0x3e 0x4b 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x1 extra data: >>>>> send: 0x3e 0x4c 0x0 0x0 0x0 0x6 0x1 0x3 0xa4 0x10 0x0 0x1 >>>>> recv: 0x3e 0x4c 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x55 0xbb extra data: >>>>> send: 0x1 0xa 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0xd 0x0 0x1 >>>>> recv: 0x1 0xa 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x0 extra data: >>>>> send: 0x1 0xb 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0xe 0x0 0x1 >>>>> recv: 0x1 0xa 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x0 extra data:

mrestorff, 2025-11-29:
Marstek changed some of the Modbus registers from V122 to V139, as far as I know. You can request the most recent firmware version through their support.

filipvh, 2025-11-29:
> Yes, its because of the old bms, ask in the chat to marstek for am update.

did that already, thx for info

ViperRNMC, 2025-12-09:
Nice, I'll add it to the readme.

sker65, 2025-12-14:
I think 33000 (total charge) and 33002 (total discharge) are not working in v3. I have a brand new set of marstek venus e gen3 with EW11 modbus bridge and always get the same value for both. no matter if i charge or discharge.

sker65, 2025-12-14:
BMS-Version 106
EMS-Version 135
CommVer: 202409090159

ViperRNMC, 2025-12-14:
Ask for an update, i see in this thread multiple times firmware changed the way modbus works on the v3

DerAutomatiker, 2025-12-19:
> > I just got the modbus integration to work on the V3 EMS 144 VNS 116 BMS 110 using the normal network connection (ethernet not the wifi) connecting to the IP on port 502.
> > For the V3 with a wired ethernet connection using a modbus adaptor is not necessary anymore :D
> 
> Hi, which communication model version you got? I have EMS (?) 149 and communication 202409.... and I don't even see open 502 port neither on LAN, nor on wifi.

EMS 149 would be a Venus V2 and not V3?!

I had communication Module 202409 and it worked on the V3 and now got communication Module 2025 and it's working also.

ViperRNMC, 2025-12-27:
[image] under devices 3 dots. Yes you can help, see in first post the missing registers.

sker65, 2025-12-27:
When I used a enhanced logging to see what value is coming from which device the debug log just confirms what I see in HA: modbus register (33002 in this example) is just delivering the same value, even though all three batteries are discharged at the same rate in the test period:
`
grep "33002" ./home-assistant_marstek_modbus_2025-12-27T20-29-25.096Z.log
2025-12-27 21:21:40.670 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Requesting register 33002 (0x80EA) from 'total_discharging_energy' for sensor '192.168.178.91' (type: int32, count: 2)
2025-12-27 21:21:40.670 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Received data from '192.168.178.91' for register 33002 (0x80EA): [0, 507]
2025-12-27 21:21:40.671 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Updated sensor 'total_discharging_energy': register=33002, value=507, scale=0.01, unit=kWh
2025-12-27 21:21:42.702 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Requesting register 33002 (0x80EA) from 'total_discharging_energy' for sensor '192.168.178.92' (type: int32, count: 2)
2025-12-27 21:21:42.702 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Received data from '192.168.178.92' for register 33002 (0x80EA): [0, 511]
2025-12-27 21:21:42.702 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Updated sensor 'total_discharging_energy': register=33002, value=511, scale=0.01, unit=kWh
2025-12-27 21:22:03.893 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Requesting register 33002 (0x80EA) from 'total_discharging_energy' for sensor '192.168.178.88' (type: int32, count: 2)
2025-12-27 21:22:03.893 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Received data from '192.168.178.88' for register 33002 (0x80EA): [0, 3368]
2025-12-27 21:22:03.893 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Updated sensor 'total_discharging_energy': register=33002, value=3368, scale=0.01, unit=kWh
2025-12-27 21:22:49.669 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Requesting register 33002 (0x80EA) from 'total_discharging_energy' for sensor '192.168.178.91' (type: int32, count: 2)
2025-12-27 21:22:49.669 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Received data from '192.168.178.91' for register 33002 (0x80EA): [0, 507]
2025-12-27 21:22:49.669 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Updated sensor 'total_discharging_energy': register=33002, value=507, scale=0.01, unit=kWh
2025-12-27 21:22:52.310 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Requesting register 33002 (0x80EA) from 'total_discharging_energy' for sensor '192.168.178.92' (type: int32, count: 2)
2025-12-27 21:22:52.310 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Received data from '192.168.178.92' for register 33002 (0x80EA): [0, 511]
2025-12-27 21:22:52.310 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Updated sensor 'total_discharging_energy': register=33002, value=511, scale=0.01, unit=kWh
2025-12-27 21:26:17.667 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Requesting register 33002 (0x80EA) from 'total_discharging_energy' for sensor '192.168.178.91' (type: int32, count: 2)
2025-12-27 21:26:17.667 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Received data from '192.168.178.91' for register 33002 (0x80EA): [0, 507]
2025-12-27 21:26:17.667 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Updated sensor 'total_discharging_energy': register=33002, value=507, scale=0.01, unit=kWh
2025-12-27 21:26:21.950 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Requesting register 33002 (0x80EA) from 'total_discharging_energy' for sensor '192.168.178.92' (type: int32, count: 2)
2025-12-27 21:26:21.950 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Received data from '192.168.178.92' for register 33002 (0x80EA): [0, 511]
2025-12-27 21:26:21.951 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Updated sensor 'total_discharging_energy': register=33002, value=511, scale=0.01, unit=kWh
2025-12-27 21:26:37.902 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Requesting register 33002 (0x80EA) from 'total_discharging_energy' for sensor '192.168.178.88' (type: int32, count: 2)
2025-12-27 21:26:37.902 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Received data from '192.168.178.88' for register 33002 (0x80EA): [0, 3369]
2025-12-27 21:26:37.902 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Updated sensor 'total_discharging_energy': register=33002, value=3369, scale=0.01, unit=kWh
2025-12-27 21:27:27.660 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Requesting register 33002 (0x80EA) from 'total_discharging_energy' for sensor '192.168.178.91' (type: int32, count: 2)
2025-12-27 21:27:27.661 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Received data from '192.168.178.91' for register 33002 (0x80EA): [0, 507]
[... cut ...]
2025-12-27 21:28:40.963 DEBUG (MainThread) [custom_components.marstek_modbus.helpers.modbus_client] Received data from '192.168.178.92' for register 33002 (0x80EA): [0, 511]
2025-12-27 21:28:40.963 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Updated sensor 'total_discharging_energy': register=33002, value=511, scale=0.01, unit=kWh
`
Okay not so easy to see, but there are 3 devices 192.168.178.88, 192.168.178.91 and 192.168.178.92. While the value from 192.168.178.88 is increasing the other two stay at 507 / 511.

ViperRNMC, 2025-12-27:
Thank you for the confirmation. This really does seem like a firmware issue. Perhaps you can report this via the Marstek app so they can fix it.

Polarium37, 2025-12-29:
Hello. 
I have just noticed a communication problem between my two Venu 3 batteries and the integration: ‘connection may be slow or incorrect’.
I am using Elfin EE11A and everything was working perfectly a few days ago.
I suspect this may be due to the EMS update: 145 (currently EMS: 145 / BMS: 110 / VMS: 116).
I restarted the batteries and HAOS, but the issue persists.
Thank you for your assistance.

`2025-12-29 09:58:03.609 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Dependency key 'ac_power'
2025-12-29 09:58:03.609 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Dependency key 'battery_soc'
2025-12-29 09:58:03.609 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Dependency key 'total_discharging_energy'
2025-12-29 09:58:03.609 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Dependency key 'total_monthly_discharging_energy'
2025-12-29 09:58:03.610 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Skipping sensor 'device_name', last update 146.2s ago (180s)
2025-12-29 09:58:03.610 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Skipping sensor 'bms_version', last update 146.2s ago (180s)
2025-12-29 09:58:03.610 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Skipping sensor 'vms_version', last update 146.2s ago (180s)
2025-12-29 09:58:03.610 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Skipping sensor 'ems_version', last update 146.2s ago (180s)
2025-12-29 09:58:03.610 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Skipping sensor 'comm_module_firmware', last update 146.2s ago (180s)
2025-12-29 09:58:03.610 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Skipping sensor 'mac_address', last update 146.2s ago (180s)
2025-12-29 09:58:03.841 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Coordinator poll tick at 2025-12-29T08:58:03.841009+00:00
2025-12-29 09:58:03.841 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Dependency key 'total_charging_energy'
2025-12-29 09:58:03.841 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Dependency key 'total_monthly_charging_energy'
2025-12-29 09:58:03.841 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Dependency key 'battery_total_energy'
2025-12-29 09:58:03.841 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Dependency key 'battery_power'
2025-12-29 09:58:03.841 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Dependency key 'ac_power'
2025-12-29 09:58:03.841 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Dependency key 'battery_soc'
2025-12-29 09:58:03.841 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Dependency key 'total_discharging_energy'
2025-12-29 09:58:03.841 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Dependency key 'total_monthly_discharging_energy'
2025-12-29 09:58:03.841 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Skipping sensor 'device_name', last update 146.5s ago (180s)
2025-12-29 09:58:03.842 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Skipping sensor 'bms_version', last update 146.5s ago (180s)
2025-12-29 09:58:03.842 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Skipping sensor 'vms_version', last update 146.5s ago (180s)
2025-12-29 09:58:03.842 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Skipping sensor 'ems_version', last update 146.5s ago (180s)
2025-12-29 09:58:03.842 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Skipping sensor 'comm_module_firmware', last update 146.5s ago (180s)
2025-12-29 09:58:03.842 DEBUG (MainThread) [custom_components.marstek_modbus.coordinator] Skipping sensor 'mac_address', last update 146.5s ago (180s)
2025-12-29 09:58:13.693 WARNING (MainThread) [custom_components.marstek_modbus.coordinator] Timeout reading sensor 'battery_soc' at register 37005 - connection may be slow or incorrect
2025-12-29 09:58:13.925 WARNING (MainThread) [custom_components.marstek_modbus.coordinator] Timeout reading sensor 'battery_soc' at register 37005 - connection may be slow or incorrect
2025-12-29 09:58:23.775 WARNING (MainThread) [custom_components.marstek_modbus.coordinator] Timeout reading sensor 'battery_total_energy' at register 32105 - connection may be slow or incorrect
2025-12-29 09:58:24.008 WARNING (MainThread) [custom_components.marstek_modbus.coordinator] Timeout reading sensor 'battery_total_energy' at register 32105 - connection may be slow or incorrect
2025-12-29 09:58:33.856 WARNING (MainThread) [custom_components.marstek_modbus.coordinator] Timeout reading sensor 'battery_voltage' at register 30100 - connection may be slow or incorrect
2025-12-29 09:58:34.090 WARNING (MainThread) [custom_components.marstek_modbus.coordinator] Timeout reading sensor 'battery_voltage' at register 30100 - connection may be slow or incorrect
2025-12-29 09:58:43.938 WARNING (MainThread) [custom_components.marstek_modbus.coordinator] Timeout reading sensor 'battery_current' at register 30101 - connection may be slow or incorrect
2025-12-29 09:58:44.172 WARNING (MainThread) [custom_components.marstek_modbus.coordinator] Timeout reading sensor 'battery_current' at register 30101 - connection may be slow or incorrect
[... cut ...]
2025-12-29 10:12:52.765 DEBUG (MainThread) [custom_components.marstek_modbus.sensor] Calculated value for round_trip_efficiency_monthly: 72.9 (input values: {'charge': 53.09, 'discharge': 38.69})
2025-12-29 10:12:52.765 DEBUG (MainThread) [custom_components.marstek_modbus.sensor] Calculated value for conversion_efficiency: 0.0 (input values: {'battery_power': -4.0, 'ac_power': 0.0})
2025-12-29 10:12:52.765 DEBUG (MainThread) [custom_components.marstek_modbus.sensor] Calculated value for stored_energy: 4.35 (input values: {'soc': 85.0, 'capacity': 5.12})`